theopolis/uefi-firmware-parser #19: Python 3 Support

kupiakos, 2015-05-04:
Is Python 3 support planned for this project? It didn't successfully build on Python 3.

theopolis, 2015-05-04:
I wasn't planned. But if python 3 would be useful for you I can try this week. Is there 3-only code (or a specific project) you'd like to integrate with?

kupiakos, 2015-05-04:
I'm researching what sorts of frameworks to use for an OS-deployment project, and manipulating UEFI properly is a pretty important part of it. Nothing specific yet. I'm developing it "for the future" so I'm strictly keeping it on Python 3.

I was also planning on using it for basic deployment scripts, but I don't want to overburden you if Python 3 support would be too difficult.

theopolis, 2015-05-04:
Cool! Shouldn't take too long, let me get back to you tonight.

theopolis, 2015-05-14:
Going to take some time getting the bytes/unicode/str types straight but it's getting there (#22).

kupiakos, 2015-05-14:
The library [Six](https://pythonhosted.org/six/) might help, depending on how complicated it is.

theopolis, 2020-05-20:
Generally finished for the `uefi_firmware` module, some of the scripts still use Python 2 APIs.

closed 2020-05-20 by theopolis