theopolis/uefi-firmware-parser #31: Dell PFSSection format difference for Latitude E5X70/Precision 3510 firmware

theopolis, 2016-01-28:
@flameeyes found that Latitude E5X70/Precision 3510 firmware is not parsed correctly by the module.

The object is a DellPFS (using `fv_parser.py`'s format-discovery:
```
DellPFS: spec 0x1 size 0x1683f1f (23609119 bytes)
  Dell PFSSection: f6baa1ac-39bc-4066-92f3-2471d1e7c5c6 spec 1 ts 542002766 type 1 version 3 size 0x92848c (9602188 bytes)
    RawObject: size= 9601304 
  Dell PFSSection: 3e3df348-70a6-4a5a-8b8f-a2d2f66dd02c spec 1 ts 538988097 type 10 version 4 size 0x40148 (262472 bytes)
    RawObject: size= 262144 
  Dell PFSSection: d3eb2492-138b-484c-a7d9-c6d860c9ab91 spec 1 ts 538988097 type 33 version 7 size 0x2148 (8520 bytes)
    RawObject: size= 8192 
  Dell PFSSection: dc090eb9-0686-4ba1-8b97-2ec7c9bb0279 spec 1 ts 1313754702 type 11 version 0 size 0x685148 (6836552 bytes)
    RawObject: size= 6836224 
  Dell PFSSection: 7439ed9e-70d3-4b65-9e33-1963a7ad3c37 spec 1 ts 1313754702 type 11 version 0 size 0x685148 (6836552 bytes)
    RawObject: size= 6836224 
  Dell PFSSection: 3fc7228d-d3f7-47af-b8bc-dabcb41252ad spec 1 ts 542002766 type 1 version 0 size 0x1b8 (440 bytes)
    RawObject: size= 112 
  Dell PFSSection: f4dd8e34-3011-40fe-a0b0-6d1fa21f2371 spec 1 ts 1313754702 type 0 version 0 size 0xad1 (2769 bytes)
    RawObject: size= 2441 
  Dell PFSSection: 2d20fc12-bfb0-4580-99cc-758c8ba180cf spec 1 ts 542002766 type 1 version 1 size 0xe344 (58180 bytes)
    RawObject: size= 57852 
  Dell PFSSection: 233ae3fb-da68-4fd4-92cb-a6229a611d6f spec 1 ts 1313754702 type 1 version 0 size 0x1b8 (440 bytes)
    RawObject: size= 112 
  Dell PFSSection: 4d583fd3-f80e-4055-a145-9bec16cb33b0 spec 1 ts 1313754702 type 1 version 0 size 0x3ee (1006 bytes)
```

The `RawObject`s here are embedded `PFSFile`s, there is no handling of recursive `PFSFile`s to begin with, ha! When trying to parse the `RawObject`s with a secondary call to parsing there is unexpected behavior:

```
DellPFS: spec 0x1 size 0x9280f8 (9601272 bytes)
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
    RawObject: size= 66120 
  Dell PFSSection: 59b3e2f6-4e42-41f3-b1f4-446a84bfc6d0 spec 1 ts 0 type 0 version 0 size 0x104b5 (66741 bytes)
[....]
```

The embedded sections seem to violate the expected section header:
```py
spec, ts, ctype, version, _u1 = struct.unpack("<IIhh4s", hdr[16:32])
```

A quick peek at the beginning:

```
00000000  f6 e2 b3 59 42 4e f3 41  b1 f4 44 6a 84 bf c6 d0  |...YBN.A..Dj....|
00000010  01 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 00 00 00 00  48 02 01 00 0c 02 00 00  |........H.......|
00000030  19 00 00 00 00 00 00 00  52 cd a4 4f ff ff ff ff  |........R..O....|
00000040  1e 9d 38 7e 05 b6 56 1d  02 00 00 00 44 65 6c 6c  |..8~..V.....Dell|
00000050  58 37 00 00 00 00 00 00  00 00 00 00 05 00 00 00  |X7..............|
00000060  02 00 00 00 18 02 00 00  00 00 01 00 00 00 37 58  |..............7X|
00000070  01 00 00 00 00 00 00 00  01 00 00 00 00 00 00 00  |................|
00000080  51 00 00 00 00 00 74 00  51 00 01 00 00 10 00 00  |Q.....t.Q.......|
00000090  51 00 02 00 00 00 01 00  51 00 03 00 00 00 00 00  |Q.......Q.......|
000000a0  55 00 00 00 00 00 00 00  53 00 00 00 00 00 00 00  |U.......S.......|
000000b0  51 00 04 00 00 00 00 00  51 00 05 00 00 00 00 00  |Q.......Q.......|
000000c0  51 00 06 00 00 00 00 00  51 00 07 00 00 00 00 00  |Q.......Q.......|
000000d0  51 00 08 00 03 00 00 00  51 00 09 00 00 00 00 00  |Q.......Q.......|
000000e0  55 00 00 00 16 00 00 00  53 00 01 00 00 00 00 00  |U.......S.......|
000000f0  13 00 01 00 01 00 00 00  60 00 03 01 00 00 00 00  |........`.......|
00000100  45 00 03 00 04 00 00 00  31 00 03 00 04 00 00 00  |E.......1.......|
00000110  54 00 00 03 00 00 00 00  53 00 01 00 00 00 00 00  |T.......S.......|
00000120  13 00 01 00 02 00 00 00  61 00 03 01 00 00 00 00  |........a.......|
00000130  41 00 03 00 ff 7f 00 00  45 00 03 00 0c 00 00 00  |A.......E.......|
00000140  30 00 03 00 00 00 00 00  54 00 00 03 00 00 00 00  |0.......T.......|
00000150  12 00 01 00 01 00 00 00  b0 00 09 00 00 00 00 00  |................|
00000160  71 00 09 00 00 00 00 00  92 00 00 00 2f 00 00 00  |q.........../...|
00000170  76 00 00 01 01 00 00 00  91 00 00 00 29 00 00 00  |v...........)...|
00000180  14 00 00 00 00 00 00 00  b0 00 09 00 00 00 00 00  |................|
00000190  71 00 09 00 00 00 00 00  92 00 00 00 33 00 00 00  |q...........3...|
000001a0  10 00 00 00 01 00 00 00  b0 00 09 00 00 00 00 00  |................|
000001b0  71 00 09 00 00 00 00 00  92 00 00 00 37 00 00 00  |q...........7...|
000001c0  30 00 04 01 00 00 00 00  70 00 04 02 00 00 00 00  |0.......p.......|
000001d0  94 00 00 00 41 00 00 00  34 00 00 01 00 00 00 00  |....A...4.......|
000001e0  32 00 00 01 00 00 00 00  90 00 00 00 1b 00 00 00  |2...............|
000001f0  31 00 05 00 01 00 00 00  70 00 05 08 00 00 00 00  |1.......p.......|
00000200  94 00 00 00 3b 00 00 00  90 00 00 00 1b 00 00 00  |....;...........|
00000210  31 00 06 00 01 00 00 00  70 00 06 08 00 00 00 00  |1.......p.......|
00000220  94 00 00 00 3d 00 00 00  90 00 00 00 21 00 00 00  |....=.......!...|
00000230  31 00 07 00 01 00 00 00  70 00 07 08 00 00 00 00  |1.......p.......|
00000240  94 00 00 00 3f 00 00 00  90 00 00 00 25 00 00 00  |....?.......%...|
00000250  51 00 0f 00 01 00 00 00  90 00 00 00 42 00 00 00  |Q...........B...|
00000260  51 00 0f 00 02 00 00 00  90 00 00 00 42 00 00 00  |Q...........B...|
00000270  51 00 0f 00 03 00 00 00  90 00 00 00 42 00 00 00  |Q...........B...|
00000280  51 00 0f 00 00 00 00 00  ff 00 00 00 00 00 00 00  |Q...............|
00000290  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000002a0  78 e5 8c 8c 3d 8a 1c 4f  99 35 89 61 85 c3 2d d3  |x...=..O.5.a..-.|
000002b0  00 00 1c 00 00 00 00 00  5f 46 56 48 ff fe 10 00  |........_FVH....|
````

theopolis, 2016-01-28:
I'm guessing the embedded sections are treated as chunks. If we strip the first bit from each (not sure how to parse this yet) then append the remaining, we should get a reconstructed volume. There will be several volumes within, as is normal. /cc @NikolajSchlej, @flameeyes

I'm working on parsing this new 'version' of the PFS structures and the embedded header they've added. Will report back here when finished.

Flameeyes, 2016-01-28:
Uhm wait, is this for the X99-S-ASUS-2101.CAP (that VirusTotal marked as malware) or for the E6510 (that I uploaded as well but seemed parsed fine?)

theopolis, 2016-01-28:
This is for the sample here: https://www.dropbox.com/s/btn25qwijgaez0q/LatitudeE5X70_Precision3510_1.3.6.exe_decompressed.hdr.zip?dl=0

NikolajSchlej, 2016-01-28:
@theopolis, interesting format indeed. 
Could you please make a writeup on the format after the analysis is done, I will then write a small standalone tool to convert that EXE files into the normal BIOS region image, like [this](https://github.com/LongSoft/InsydeImageExtractor).

lordkag, 2016-01-28:
I might be able to help with this one. I had this added to UEFI_Strip for a while, as I mentioned in [this issue](https://github.com/LongSoft/UEFITool/issues/32), first report. I haven't posted a public release in a long time, because there wasn't much of an user base. However, I've been sharing trunk versions with Sonix, Plutomaniac and Pacman. CodeRush, I haven't sent you a link, because I didn't knew if you had any need for it and didn't wanted to force you to view my horrible coding. Just because I'm a genius robot, it doesn't mean that I should code like one. Since I already planed to add a new bug report tonight to UEFITool and have some extra related questions reserved for PM, I can send you the link, if you want.

[Here is how dellhdr option looks in action](http://www.mediafire.com/view/d9brudro1ztnq1w/Dell_HDR.png) and [here is the magic](http://www.mediafire.com/download/6c3mit34o6s8dfr/Dell_HDR.rar). Please excuse the crappy code.

NikolajSchlej, 2016-01-28:
@lordkag, thanks a lot. Please have no fear to show any "crappy" code, actually, it's already very good just because it works.

lordkag, 2016-01-28:
Thanks. I will also send you a link to a new tool, which I like to call BEEF, as in BIOS Extract from Exe File. It deals with extracting the content from all these exotic executables. Apart from adding Dell_HDR [from here](http://forums.mydigitallife.info/threads/44785-I-present-you-a-tool-to-decompress-Dell-UEFI-BIOS) and your InsydeExtractor (with a few touches), I have added support for AMIUCP packaging, Dell Winflash, Dell Wistron, Dell Flash ME, Dell Server, Dell ROM/HDR, Phoenix SCT. Apparently Dell is keeping me busy.

NikolajSchlej, 2016-01-28:
@lordkag, nice, less work for me :smile: 
You should upload your tools to Guthub too, because then not only me, Sonix, PM and Pacman can use them and contribute to them, but the whole world. People are getting more and more interested in the topic, and the field lacks just about any tools (I don't speak about "good" or even "decent" ones, it's really about "any"). We need to cooperate, we need to share, we need to publish bug reports and make pull requests, only this way we can prevent constant reinvention of the wheel.
Please don't be afraid of a dirty code. The code itself has no real value and can be rewritten with ease, the algorithms, tricks, quirks - that are much more important, and the code itself can be polished later once we get all other things perfect.

theopolis, 2016-01-29:
Awesome! Great stuff @lordkag!

closed 2016-01-30 by theopolis