theopolis/uefi-firmware-parser #44: fv_parser: "--brute" not working anymore

weidenba, 2016-06-29:
Hello,
First of all: Great work... I really like your project.

Now to my issue:
Since you removed line 13 ("parsed_object.process()") in fv_parser.py I am no longer able to extract UEFI-Files  that require the "--brute" option (e.g. ASUS cap files). 
Was this line removed on purpose or is a workaround available?

If I re-add the line, everything works fine for me. 

Tracedump:
```
Firmware Volume: 8c8ce578-8a3d-4f1c-9935-896185c32dd3 attr 0x0003feff, rev 2, cksum 0xe6ae, size 0x20000 (131072 bytes)
  Firmware Volume Blocks: 
Traceback (most recent call last):
  File "fv_parser.py", line 70, in <module>
    brute_search_volumes(input_data)
  File "fv_parser.py", line 23, in brute_search_volumes
    parse_firmware_volume(data[index - 40:], name=index - 40)
  File "fv_parser.py", line 31, in parse_firmware_volume
    _process_show_extract(firmware_volume)
  File "fv_parser.py", line 13, in _process_show_extract
    parsed_object.showinfo('')
  File "build/bdist.linux-x86_64/egg/uefi_firmware/uefi.py", line 1170, in showinfo
TypeError: 'NoneType' object is not iterable

```
Greetings
 Peter

theopolis, 2016-06-29:
Thanks for the report @weidenba! I'll take care of this and add a test to make sure it doesn't happen again. I'll grab one of the ASUS updates from my repo and add a new section to the `uefi-firmware-samples` collection that is designed to use the `--brute` flag.

closed 2016-07-01 by theopolis

theopolis, 2016-07-04:
@weidenba, also, checkout the new `--superbrute` option to help with ASUS CAP updates. That option should find the embedded flash descriptor and parse a lot more.

commit 9545e045 (2016-07-01): [Fix #44] --brute and add --superbrute (#45)