theopolis/uefi-firmware-parser #48: Intel ME EFFS partition parsing

theopolis, 2016-07-04:
The EFFS partition in Intel ME containers contains structured data. Parsing would be nice, similar to the NVRAM EFI region parsing request.

skochinsky, 2017-04-26:
@theopolis check out https://www.troopers.de/downloads/troopers17/TR17_ME11_Static.pdf (last slides); you may want to contact them about this

orangecms, 2024-12-05:
Hey I'm currently reversing the EFFS / MFS for ME Gen 2 (version 6-10), i.e. what came before what PT have reversed.
I have WIP stuff in a branch in this repo, with a sketch of what I figured out so far:
https://github.com/fiedka/me_fs_rs/blob/mfs/doc/me_gen2_mfs.md
Code is here:
https://github.com/fiedka/me_fs_rs/blob/mfs/src/mfs/gen2.rs

@skochinsky it has been quite some years since you looked at it. I found nothing public on the Gen 2 FS, and in one of your presentations you said that it's complicated due to wear leveling :sweat_smile: have you gotten further?
Anyhow, I have a few samples from different devices and upgrade images, and I'll get some more dumps from devices after running for a while to see what changes over time. So far I only have pages and what I think are the actual data chunks, but I cannot make sense of how they are indexed / addressed / markes as live etc..

@theopolis I have reimplemented what PT had reversed in Rust, also in that repo, in https://github.com/fiedka/me_fs_rs/blob/mfs/src/mfs/gen3.rs - with some WIP stuff in there because it didn't fit for some additional things I started looking at. Anyhow, there is also Python code in PT's repos, though Python 2. I guess that is most useful to you then: https://github.com/ptresearch/parseMFS

I'll be happy to help out if you have any questions. :)