theopolis/uefi-firmware-parser #53: Support for Lenovo Phoenix updates

skochinsky, 2017-03-09:
This is basically two (or three) in one:
1) support for unpacking InnoSetup exes (e.g. using [innoextract](https://github.com/dscharrer/innoextract))
2) handling of .FL1/FL2 files. These usually come in two kinds:
 2a) full ROM .fds, either as-is, or packed with [BCPVPD compression](https://github.com/coreboot/bios_extract/blob/master/bcpvpd.c)
 2b) one or more regions (bios/me/ec/desc/gbe) in an image with an [`$PFH` trailer](https://gist.github.com/skochinsky/181e6e338d90bb7f2693098dc43c6d54) near the end (0x1000 aligned)
 3) [maybe] support for extracting old pre-UEFI [Phoenix FFV](https://github.com/coreboot/bios_extract/blob/master/phoenix_extract.py) filesystem.

theopolis, 2017-03-13:
The type-test detection scripts are not working for these?
```
uefi-firmware-parser --test ~/firmware/*
```

Is it possible to get my hands on examples? I'd be happy to write the support.

skochinsky, 2017-03-13:
For samples, see Thinkpad updates, e.g. this one for x220:
https://download.lenovo.com/pccbbs/mobiles/8duj28us.exe

commit aea0eaf4 (2017-04-08): [Fix #53] Add 'basic' Lenovo Phoenix (fl1) parsing
commit 81a46aa2 (2017-04-13): [Fix #53] Add 'basic' Lenovo Phoenix (fl1) parsing (#54)
closed 2017-04-13 by theopolis