theopolis/uefi-firmware-parser #59: Volume extraction directory names do not match the firmware layout

osresearch, 2018-01-09:
When running `uefi-firmware-parser --extract` the `regions/region-bios` volumes are parsed correctly, but the extracted names do not make sense.  For instance, the SecCore file with guid `1ba0062e` is in the last firmware volume, with GUID `6522280D-28F9-4131-ADC4-F40EBFA45864`, but the file is extracted into: `./regions/region-bios/volume-0/file-1ba0062e-c779-4582-8566-336ae8f78f09/file.obj`  And since this firmware has both a recovery and a normal region, both copies are assigned the same name and will overwrite each other.

Meanwhile, files stored in a compresed volume are nested, but the volume GUIDs are not used (just the compression GUID).  DxeCore with guid `d6a2cb7f` in volume `8C8CE578-8A3D-4F1C-9935-896185C32DD3` ends up extracted to: `./regions/region-bios/volume-0/file-9e21fd93-9c72-4c15-8c4b-e77f1db2d792/section0/section1/volume-ee4e5898-3914-4259-9d6e-dc7bd79403cf/file-d6a2cb7f-6a18-4e2f-b43b-9920a733700a/file.obj`.

These GUIDs don't make sense for nesting, since 9e21fd93 is generic volume image and ee4e5898 is LZMA.  That means that every compressed volume gets dumped into the same directory.

I would have expected each of the top-level firmware volume to get its own `volume-$N` directory and for the subdirectories to be named by the FV guids so that they don't clash with each other.

Here is the extraction log for reference:
[extract.txt](https://github.com/theopolis/uefi-firmware-parser/files/1616145/extract.txt)

theopolis, 2018-01-15:
Right, that is fairly nonsensical.

RockySYR, 2020-07-28:
> parse

how to parse a single volume

theopolis, 2026-11-01:
This should be fixed with https://github.com/theopolis/uefi-firmware-parser/pull/163

closed 2026-11-01 by theopolis