theopolis/uefi-firmware-parser #69: python3 seems to have dropped buffer() so script crashes

dorpvom, 2018-09-24:
Hi,

nice work to port the code to python3!

Running the script though I ran into the following exception:
```
Traceback (most recent call last):
  File "uefi-firmware-parser", line 121, in <module>
    superbrute_search(input_data)
  File "uefi-firmware-parser", line 35, in superbrute_search
    bdata = buffer(data, i)
NameError: name 'buffer' is not defined
```
This seems to stem from the fact that python has dropped `buffer()` in favor of `memoryview()` [[1]](https://docs.python.org/3/c-api/buffer.html).
When merely replacing the `buffer(data, i)` with the possibly equal `memoryview(data[i:])`, I get another exception:
```
Traceback (most recent call last):
  File "uefi-firmware-parser", line 121, in <module>
    superbrute_search(input_data)
  File "uefi-firmware-parser", line 38, in superbrute_search
    _process_show_extract(parser.parse())
  File "/usr/local/lib/python3.5/dist-packages/uefi_firmware-1.8-py3.5-linux-x86_64.egg/uefi_firmware/__init__.py", line 81, in parse
    if not self.firmware.process():
  File "/usr/local/lib/python3.5/dist-packages/uefi_firmware-1.8-py3.5-linux-x86_64.egg/uefi_firmware/flash.py", line 118, in process
    self.map = DescriptorMap(self.data[20:20 + DescriptorMap.size])
  File "/usr/local/lib/python3.5/dist-packages/uefi_firmware-1.8-py3.5-linux-x86_64.egg/uefi_firmware/flash.py", line 30, in __init__
    self.parse_structure(data, FlashDescriptorMapType)
  File "/usr/local/lib/python3.5/dist-packages/uefi_firmware-1.8-py3.5-linux-x86_64.egg/uefi_firmware/base.py", line 124, in parse_structure
    ctypes.addressof(struct_instance), struct_data, struct_length)
ctypes.ArgumentError: argument 2: <class 'TypeError'>: wrong type
```
Might be an easy fix for all this but I don't see it. A fix or a 'no-you-just-have-to-call-it-this-way' would be appreciated ^^

theopolis, 2020-05-20:
This is challenging indeed. I tried

```diff
diff --git a/bin/uefi-firmware-parser b/bin/uefi-firmware-parser
index dd786a7..216ab11 100755
--- a/bin/uefi-firmware-parser
+++ b/bin/uefi-firmware-parser
@@ -33,7 +33,7 @@ def _process_show_extract(parsed_object):
 def superbrute_search(data):
     for i in range(len(data)):
         # using buffer avoids long memory copy
-        bdata = buffer(data, i)
+        bdata = data[i:]
         parser = AutoParser(bdata, search=False)
         if parser.type() is not 'unknown':
             _process_show_extract(parser.parse())
@@ -119,7 +119,7 @@ if __name__ == "__main__":
 
         try:
             with open(file_name, 'rb') as fh:
-                input_data = fh.read()
+                input_data = memoryview(fh.read())
         except Exception as e:
             print("Error: Cannot read file (%s) (%s)." % (file_name, str(e)))
             continue
```

And this does not work either since farther down in the code we expect to run `.find` on the readonly buffer and `memoryview` does not implement a search/find API.

theopolis, 2020-06-24:
For now the script is Python3 compatible but will be much slower (back to the original slow speed) due to not having a `buffer` alternative.

closed 2020-06-24 by theopolis