theopolis/uefi-firmware-parser #71: errors on python 2.7.13 on Windows

skochinsky, 2018-10-28:
1. build of EfiCompressor.c fails with "VC++ 2008 for Python" due to missing stdint.h. A copy should be added to the `Include` directory.
2. script installed into `Scripts` should have .py extension or an .exe runner added (check how Chipsec does it)
3. Script fails:
```C:\python27-x64\Scripts>uefi-firmware-parser.py
Traceback (most recent call last):
  File "C:\python27-x64\Scripts\uefi-firmware-parser.py", line 4, in <module>
    __import__('pkg_resources').run_script('uefi-firmware==1.8', 'uefi-firmware-parser')
  File "C:\python27-x64\lib\site-packages\pkg_resources\__init__.py", line 743, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "C:\python27-x64\lib\site-packages\pkg_resources\__init__.py", line 1505, in run_script
    exec(script_code, namespace, namespace)
  File "C:\python27-x64\lib\site-packages\uefi_firmware-1.8-py2.7-win-amd64.egg\EGG-INFO\scripts\uefi-firmware-p
arser", line 10, in <module>

  File "build\bdist.win-amd64\egg\uefi_firmware\__init__.py", line 5, in <module>
  File "build\bdist.win-amd64\egg\uefi_firmware\uefi.py", line 11, in <module>
  File "build\bdist.win-amd64\egg\uefi_firmware\base.py", line 8, in <module>
  File "build\bdist.win-amd64\egg\uefi_firmware\utils.py", line 7, in <module>
ImportError: No module named builtins
```

404notfoundhard, 2019-01-24:
@skochinsky,
Try to install package "future" with pip.
I had a similar problem on ubuntu 18.04 and it helped me.

theopolis, 2020-05-20:
I added this to `setup.py` as @404notfoundhard suggested in https://github.com/theopolis/uefi-firmware-parser/pull/73/files hopefully this works now!

closed 2020-05-20 by theopolis