theopolis/uefi-firmware-parser #76: error with pip3 install macOS 10.13 & Python 3.7.5

sadponyguerillaboy, 2019-12-19:
`Collecting uefi_firmware
  Downloading https://files.pythonhosted.org/packages/ea/e0/d8e060ec233777faab2db6bf060e905b3ddeaa5be172068f5954212372f9/uefi_firmware-1.7.tar.gz (176kB)
     |████████████████████████████████| 184kB 3.0MB/s 
Building wheels for collected packages: uefi-firmware
  Building wheel for uefi-firmware (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-install-azgd60ji/uefi-firmware/setup.py'"'"'; __file__='"'"'/private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-install-azgd60ji/uefi-firmware/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-wheel-zkisps6y --python-tag cp37
       cwd: /private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-install-azgd60ji/uefi-firmware/
  Complete output (94 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.13-x86_64-3.7
  creating build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/uefi.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/me.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/__init__.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/utils.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/flash.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/pfs.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  copying uefi_firmware/base.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware
  creating build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/misc
  copying uefi_firmware/misc/__init__.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/misc
  copying uefi_firmware/misc/checker.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/misc
  creating build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/efiguids.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/efiguids_lenovo.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/efiguids_ami.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/__init__.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/efiguids_asrock.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  copying uefi_firmware/guids/efiguids_dell.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/guids
  creating build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/generator
  copying uefi_firmware/generator/uefi.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/generator
  copying uefi_firmware/generator/__init__.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/generator
  creating build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/structs
  copying uefi_firmware/structs/flash_structs.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/structs
  copying uefi_firmware/structs/__init__.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/structs
  copying uefi_firmware/structs/uefi_structs.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/structs
  copying uefi_firmware/structs/intel_me_structs.py -> build/lib.macosx-10.13-x86_64-3.7/uefi_firmware/structs
  running build_ext
  building 'uefi_firmware.efi_compressor' extension
  creating build/temp.macosx-10.13-x86_64-3.7
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression/Tiano
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression/LZMA
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression/LZMA/SDK
  creating build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression/LZMA/SDK/C
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -Iuefi_firmware/compression/Include -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c uefi_firmware/compression/EfiCompressor.c -o build/temp.macosx-10.13-x86_64-3.7/uefi_firmware/compression/EfiCompressor.o
  uefi_firmware/compression/EfiCompressor.c:109:44: error: no member named 'bf_getsegcount' in 'PyBufferProcs'
    SegNum = SrcData->ob_type->tp_as_buffer->bf_getsegcount((PyObject *)SrcData, NULL);
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
  uefi_firmware/compression/EfiCompressor.c:115:43: error: no member named 'bf_getreadbuffer' in 'PyBufferProcs'; did you mean 'bf_getbuffer'?
      Len = SrcData->ob_type->tp_as_buffer->bf_getreadbuffer((PyObject *)SrcData, Index, &BufSeg);
                                            ^~~~~~~~~~~~~~~~
                                            bf_getbuffer
  /usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/include/python3.7m/object.h:314:20: note: 'bf_getbuffer' declared here
       getbufferproc bf_getbuffer;
                     ^
  uefi_firmware/compression/EfiCompressor.c:115:59: error: called object type '<dependent type>' is not a function or function pointer
      Len = SrcData->ob_type->tp_as_buffer->bf_getreadbuffer((PyObject *)SrcData, Index, &BufSeg);
                                                            ^
  uefi_firmware/compression/EfiCompressor.c:159:42: error: no member named 'bf_getreadbuffer' in 'PyBufferProcs'; did you mean 'bf_getbuffer'?
        || SrcData->ob_type->tp_as_buffer->bf_getreadbuffer == NULL
                                           ^~~~~~~~~~~~~~~~
                                           bf_getbuffer
  /usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/include/python3.7m/object.h:314:20: note: 'bf_getbuffer' declared here
       getbufferproc bf_getbuffer;
                     ^
[... 114 lines cut ...]
          || SrcData->ob_type->tp_as_buffer->bf_getreadbuffer == NULL
                                             ^~~~~~~~~~~~~~~~
                                             bf_getbuffer
    /usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/include/python3.7m/object.h:314:20: note: 'bf_getbuffer' declared here
         getbufferproc bf_getbuffer;
                       ^
    uefi_firmware/compression/EfiCompressor.c:230:42: error: no member named 'bf_getsegcount' in 'PyBufferProcs'
          || SrcData->ob_type->tp_as_buffer->bf_getsegcount == NULL) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
    uefi_firmware/compression/EfiCompressor.c:270:10: warning: implicit declaration of function 'PyBuffer_FromMemory' is invalid in C99 [-Wimplicit-function-declaration]
      return PyBuffer_FromMemory(DstBuf, (Py_ssize_t)DstDataSize);
             ^
    uefi_firmware/compression/EfiCompressor.c:270:10: warning: incompatible integer to pointer conversion returning 'int' from a function with result type 'PyObject *' (aka 'struct _object *') [-Wint-conversion]
      return PyBuffer_FromMemory(DstBuf, (Py_ssize_t)DstDataSize);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    uefi_firmware/compression/EfiCompressor.c:255:14: warning: comparison of integers of different signs: 'UINT32' (aka 'unsigned int') and 'int' [-Wsign-compare]
      if (Status == EFI_BUFFER_TOO_SMALL) {
          ~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~
    uefi_firmware/compression/EfiCompressor.c:378:3: warning: implicit declaration of function 'Py_InitModule3' is invalid in C99 [-Wimplicit-function-declaration]
      Py_InitModule3("efi_compressor", EfiCompressor_Funcs, "Various EFI Compression Algorithms Extension Module");
      ^
    6 warnings and 7 errors generated.
[... cut ...] ware/setup.py'"'"'; __file__='"'"'/private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-install-azgd60ji/uefi-firmware/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/4k/j439323d7cqb8tg2zcscw03c0000gn/T/pip-record-c8_eqiqf/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.`

theopolis, 2019-12-22:
Hi @sadponyguerillaboy, thanks for the bug report. I am not actively developing this module so I will not have time to fix the issue. If you have a fix yourself, perhaps some logic to use the new libpython APIs I am happy to review and merge.

Also, my guess is this issue should be caught if the TravisCI tests include 3.7 in addition to what it does now, which is 3.6.

screwer, 2020-03-23:
Hi @sadponyguerillaboy. Try to install from github, not via pip. (check read.me instructions on repo main page)

sadponyguerillaboy, 2020-03-23:
@screwer thanks, for the update. But already got it working with python 2 on linux a while ago. But will give it a go, will be handy to have.

sadponyguerillaboy, 2020-03-25:
@screwer, the github installation worked like a charm!

theopolis, 2020-05-20:
I added testing/Travis support for 3.7 and pushed the latest versions to Pypi, hopefully this fixes the issue.

closed 2020-05-20 by theopolis