theopolis/uefi-firmware-parser #77: Extracting blobs/ROM image from original Lenovo FL2 files

tlaurion, 2020-03-13:
@theopolis : As specified [here](https://github.com/osresearch/heads/issues/307#issuecomment-598363617), trying to do a PoC to be able to extract ME related regions from original Lenovo bios updates (hopefully 8MB rom image) from Lenovo FL2 file, and nothing extracted matches me_cleaner expected file format:

`wget https://download.lenovo.com/pccbbs/mobiles/g2uj31us.exe && innoextract ~/g2uj31us.exe && echo -e "\n\n EXTRACTING FL1\n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL1 -e && echo -e "\n\n\n EXTRACTING FL2 \n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL2 -e --superbrute && echo -e "\n\n\n Testing me_cleaner against all extracted files.... output given only on successful files... \n\n\n" && find ./pfheader* ./capsule-Capsule/ ./volume-0/ -type f | while read filename; do python ~/me_cleaner/me_cleaner.py $filename &>/dev/null; if [ $? -eq 0 ]; then echo $filename; fi; done;`

Also, FL2 regions need to be extracted with --superbrute. Any advice?

tlaurion, 2020-03-13:
I would have thought https://github.com/theopolis/uefi-firmware-parser/issues/53 permitted this.

theopolis, 2020-03-15:
It looks like you found a way forward in discussion on the `heads` project. I'm on leave right now but I can take a look at adding (or fixing) extraction within `uefi-firmware-parser` in a few months when I return. 😄

tlaurion, 2020-03-15:
> It looks like you found a way forward in discussion on the `heads` project. I'm on leave right now but I can take a look at adding (or fixing) extraction within `uefi-firmware-parser` in a few months when I return. smile

@theopolis: yeah, not perfect though. Would be better if it was possible to extract the ME ROM region correctly and apply me_cleaner directly on the full image. IFD descripto still has to be extracted from flashed image.

theopolis, 2020-03-15:
Ok, I'll read though all of the dev/debugging in the related issues and try to come up with the step-by-step for extracting and see if it can be done with this parser/extractor.

tlaurion, 2022-01-22:
You can read https://github.com/chipsec/chipsec/issues/870#issuecomment-1019010647 and close this issue!

closed 2026-11-01 by theopolis