theopolis/uefi-firmware-parser #89: run :$ uefi-firmware-parser -O rom.bin  and run successfully ,i can see a rom.bin folder ,but the folder is empty

RockySYR, 2020-07-28:
i try other methods 
run: $ uefi-firmware-parser -o OUTPUT rom.bin
but also did not generate the parse file ,only print parse result

RockySYR, 2020-07-28:
> 我尝试其他方法
> 运行： $ uefi - 固件 - 解析器
> - o 输出 rom. bin， 但也没有生成解析文件， 只打印解析结果

RockySYR, 2020-07-28:
> i try other methods
> run: $ uefi-firmware-parser -o OUTPUT rom.bin
> but also did not generate the parse file ,only print parse result
> @anatol @epitron @thops @andreiw @theopolis

wangzitiansky, 2020-07-28:
I met the problem too

RockySYR, 2020-07-28:
> I met the problem too

my god ，how to solve this problem

Krazubu, 2020-08-31:
Same issue here, it seems to understand the content as it prints lots of UUIDs, with detail about the type and even gives the file name, but the output folder is empty.
That's under OSX 10.15.6.

theopolis, 2026-11-01:
Hello, please let me know if you are still experiencing this issue on the most recent release.

closed 2026-11-01 by theopolis