theopolis/uefi-firmware-parser #93: JSON output support

orangecms, 2020-10-11:
For further processing, e.g., visualization, it would be great to have JSON output support.

The current code is tailored towards line by line output with indentation and would need some refactoring.
I started looking into it. :)

theopolis, 2020-10-11:
I agree JSON would be a much better alternative. I’m happy to merge an implementation if you send a PR!

orangecms, 2020-10-14:
I am struggling with Python... how do I get the script in `bin/` to properly load everything from the project directory? Usually I would just set `PYTHONPATH=$(pwd)` and it's all good. Here, I am getting a headache with this error:

```
/home/dan/firmware/uefi-firmware-parser
Traceback (most recent call last):
  File "./usr/bin/uefi-firmware-parser", line 11, in <module>
    from uefi_firmware.uefi import *
  File "/home/dan/firmware/uefi-firmware-parser/uefi_firmware/__init__.py", line 5, in <module>
    from . import uefi
  File "/home/dan/firmware/uefi-firmware-parser/uefi_firmware/uefi.py", line 17, in <module>
    from uefi_firmware import efi_compressor
ImportError: cannot import name 'efi_compressor' from partially initialized module 'uefi_firmware' (most likely due to a circular import) (/home/dan/firmware/uefi-firmware-parser/uefi_firmware/__init__.py)
```

When trying

```sh
export PYTHONPATH=$(pwd)
echo $PYTHONPATH # this prints out the correct path

./bin/uefi-firmware-parser 
```

I do notice that the compressor thing here is invoking C code at some point. I have search for the error around circular imports and what not, but cannot find any hint towards solving this. :/ How do you do it?

theopolis, 2020-10-19:
Hey @orangecms, on macOS I usually do something like:

```bash
$ mkdir /tmp/uefi-dev && cd /tmp/uefi-dev
$ python3 -m venv /tmp/uefi-dev
$ source ./bin/activate
$ git clone https://github.com/theopolis/uefi-firmware-parser
$ cd uefi-firmware-parser
$ python ./setup.py develop
$ which uefi-firmware-parser
/tmp/uefi-dev/bin/uefi-firmware-parser
```

orangecms, 2020-10-19:
I see, the point I was missing is the `python ./setup.py develop` step. Thank you! :)

PR #94 (2020-11-20): json output
- WIP: JSON output support
- add json method to FirmwareVolume

to resolve #93
closed 2020-11-20 by theopolis