theopolis/uefi-firmware-parser #101: contrib/dell_extract_hdr: `UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 2: invalid start byte`

paulmenzel, 2021-05-13:
Trying to extract HDR from the [Dell OptiPlex 5055 1.1.25 firmware update file](https://dl.dell.com/FOLDER07314150M/1/OptiPlex%205055%20Ryzen%20CPU_1.1.25.exe) fails with the error below:

    $ ~/src/uefi-firmware-parser/scripts/contrib/dell_extract_hdr.py ~/OptiPlex\ 5055\ Ryzen\ CPU_1.1.25.exe
    Traceback (most recent call last):
      File "/home/joey/src/uefi-firmware-parser/scripts/contrib/dell_extract_hdr.py", line 38, in <module>
        data = fh.read()
      File "/pkg/python-3.8.9-1/lib/python3.8/codecs.py", line 322, in decode
        (result, consumed) = self._buffer_decode(data, self.errors, final)
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 2: invalid start byte

paulmenzel, 2021-05-13:
$ binwalk ~/OptiPlex\ 5055\ Ryzen\ CPU_1.1.25.exe

    DECIMAL       HEXADECIMAL     DESCRIPTION
    --------------------------------------------------------------------------------
    0             0x0             Microsoft executable, portable (PE)
    266024        0x40F28         CRC32 polynomial table, little endian
    270120        0x41F28         CRC32 polynomial table, big endian
    274399        0x42FDF         Copyright string: "Copyright 1995-2010 Mark Adler "
    304854        0x4A6D6         Copyright string: "Copyright 2009-2019"
    305342        0x4A8BE         Copyright string: "Copyright 2009-2019"
    502800        0x7AC10         Zlib compressed data, default compression
    6502200       0x633738        Object signature in DER format (PKCS header length: 4, sequence length: 8662
    6502341       0x6337C5        Certificate in DER format (x509 v3), header length: 4, sequence length: 1277
    6503622       0x633CC6        Certificate in DER format (x509 v3), header length: 4, sequence length: 1309
    6505590       0x634476        Object signature in DER format (PKCS header length: 4, sequence length: 5272
    6505751       0x634517        Certificate in DER format (x509 v3), header length: 4, sequence length: 1066
    6506821       0x634945        Certificate in DER format (x509 v3), header length: 4, sequence length: 1299
    6508124       0x634E5C        Certificate in DER format (x509 v3), header length: 4, sequence length: 1548

paulmenzel, 2021-05-13:
After `binwalk --extract ~/OptiPlex\ 5055\ Ryzen\ CPU_1.1.25.exe`, the directory `_OptiPlex 5055 Ryzen CPU_1.1.25.exe.extracted` contains the (uncompressed) 16784441 bytes (≈ 16 MB) file `7AC10`, which [UEFITool](https://github.com/LongSoft/UEFITool) is able to open.

theopolis, 2026-11-01:
This should fix it: https://github.com/theopolis/uefi-firmware-parser/pull/162

closed 2026-11-01 by theopolis