theopolis/uefi-firmware-parser #116: PyPI distribution missing source and wheels for python >3.9

twizmwazin, 2023-03-14:
The PyPI distribution is missing a source dist and wheels for python >3.9. This means that to use it in a newer python version the package must be installed from git. Adding a source distribution would allow it to be built locally by the user when a pre-built package is not able to be provided.

yeggor, 2023-07-14:
the other problem is that PyPI distribution for `uefi-firmware 1.10` contains only `*x86_64.whl`, so `v1.10` can not be installed on arm64 machines. @theopolis is is possible to add source distribution as well like it was done for older versions?

nashi5566, 2023-07-31:
Hi, I also faced same problem and followed this issue, trying to install the package on amd64 Linux Docker but failed.
My environment is:
```
host os: macOS Ventura (intel Core i5)
guest os: ubuntu 22.04 jellyfish (amd64)
python 3.10, pip 22.0.2
```

theopolis, 2023-07-31:
I'll try to work on the packaging this week. The problems come from my lack of familiarity with Python packaging.

twizmwazin, 2023-07-31:
@theopolis If you want I can PR a GitHub action to run on tagged releases that will build it and push it to PyPI, or just a simple script if you prefer that.

theopolis, 2023-07-31:
Wow, that GitHub action workflow would be outstanding!

yeggor, 2023-07-31:
> I'll try to work on the packaging this week. The problems come from my lack of familiarity with Python packaging.

I usually do something like this:

```
uefi-firmware-parser$ python setup.py sdist bdist_wheel # this command will generate both whl and source distribution

uefi-firmware-parser$ ls dist
uefi_firmware-1.10-cp310-cp310-macosx_13_0_arm64.whl uefi_firmware-1.10.tar.gz

uefi-firmware-parser$ twine upload -u __token__ dist/*
```

yeggor, 2023-07-31:
> Only build the binary distribution for the platform/abi/architecture on the local machine?

yep, it will. But if you will publish source distribution as well (`python setup.py sdist`) on the other machines it can be compiled during installation. The only disadvantage is that developer tools will be required for installation (which not all users will like).

> I wonder if there's a pure-python implementation of the EFI compressor module that would make an any distribution possible?

I didn't find one.
It also was discussed here: https://github.com/platomav/BIOSUtilities/issues/16.

@platomav wrote this:

> For me, an interesting alternative would have been a pure python implementation which is cross-OS compatible, but no such thing exists, so the current TianoCompress usage is reliable and generally good enough. Similarly, for LZNT1 compression, I use this nice [pure python module](https://pypi.org/project/lznt1/) because it is not Windows-only and does not require any external libraries/compiling. Its only issue is how slow it is, but that is an acceptable downside for the other two positives.

yeggor, 2023-07-31:
> Only build the binary distribution for the platform/abi/architecture on the local machine?

it turns out there are two options for now?
* cover all x86 platforms for `bdist_wheel` using GitHub actions with runners on ubuntu, windows and mac, and for the other platforms upload `sdist`
* implement EFI compressor module in python and distribute platform and architecture independent `bdist_wheel` (`py3-none-any.whl`)

theopolis, 2023-07-31:
I'll take a look at the complexity of the compressor implementation and see if I can write it in Python.

PR #120 (2023-08-01): Add release pipeline, fixes #116
This release pipeline uses cibuildwheel to build wheels for most common platforms and also an sdist for platforms not covered. All the build tools and publishing action are by PyPA to hopefully observe best practices. One notable thing is that this does not generate a python 2 wheel, however python 2 users will still be able to install with the provided sdist. If that is an issue I can look at adding another set of jobs for python 2 specifically using an older version of cibuildwheel.

I have tested the build jobs and the output looks good, I haven't been able  [... cut]
commit 8841b79f (2023-08-01): Add release pipeline, fixes #116 (#120)
closed 2023-08-01 by theopolis