theopolis/uefi-firmware-parser #123: Parser does not process compressed PE32 images

yeggor, 2024-01-31:
The parser does not take into account the case where we have `File -> GUID Defined Section -> PE32 image/UI`:

[image]

As far as I can see, it successfully performs the decompression, but it does not process the data with respect to parent file.

The test volume is attached:
[Volume_FFSv2_5C60F367-A505-419A-859E-2A4FF6CA6FE5.vol.zip](https://github.com/theopolis/uefi-firmware-parser/files/14118174/Volume_FFSv2_5C60F367-A505-419A-859E-2A4FF6CA6FE5.vol.zip)

NikolajSchlej, 2024-02-01:
Can you elaborate on what is the issue exactly? What do you expect to be processed and how?

yeggor, 2024-02-01:
It's probably not clear from my description, but I gave a screenshot from uefitool as an example of valid parsing.

Whereas if we run `uefi-firmware-parser` on the test volume, it will be parsed, but none of the PE images will be extracted.

Here is the log for `uefi-firmware-parser -b --verbose ../fwhunt-scan/test/Volume_FFSv2_5C60F367-A505-419A-859E-2A4FF6CA6FE5.vol`:

[log.txt](https://github.com/theopolis/uefi-firmware-parser/files/14120510/log.txt)

closed 2024-02-01 by yeggor